lwa-project/ng_digital_processor #33: NDP hard to recover after a power outage

jaycedowell, 2024-04-06:
It was a little rough getting North Arm back up after the power outage:
 * NAL didn't try to bring anything up
 * The first `NDP INI` said it was successful but it looked like all of the DRX pipelines were dead/hung.
 * Restarting the DRX pipelines helped but `snap01` wasn't sending any data.
 * Another `NDP INI` didn't fix the "no data from `snap01`" problem.
 * Power cycling the snaps and another `NDP INI` did get things working again.

Questions:
 1. Why did the DRX pipelines not start up?
 2. Why wasn't the DRX pipelines problem not detected as an error?  Did I not wait long enough?
 3. Why wasn't the the `snap01` problem detected as an error?

jaycedowell, 2024-04-09:
Something similar happened this morning (April 9) with ndp-drx-0`dying (`snap01` seems ok as does `ndp-drx-1`).  NDP did go into error because of T-engine packet loss of 50%.

GregBTaylor, 2024-04-10:
On the morning of April 9th, (around 12:20am), there was a power outage at NA.  At about 3am I did an INI on ASP and then an INI on NDP.  Both systems appeared to fully recover and reported no errors however the sky was blank.

jaycedowell, 2024-04-11:
Did you happen to look at `like_bmon.py` to see what was unhappy inside NDP?

jaycedowell, 2024-04-12:
6c3810e, 37b7ec8, and 887e0c2 are now active.

jaycedowell, 2024-04-15:
@ctaylor-physics reports that NDP was recovered with a single INI on April 14.

jaycedowell, 2024-07-12:
It seems better now if you ignore (1) how Orville sometimes doesn't automatically resume imaging (not really a NDP problem as far as I can tell) and (2) the startup packet loss thing (#30).

closed 2024-07-12 by jaycedowell